skasperski/navigation_2d #12: How to Combine Tutorial 2 and Tutorial 3 to display laser scan on top of the static map in Rviz ?

ktiwari9, 2015-08-07:
Hello there,
I am currently practicing the autonomous navigation of a p3dx robot in rviz. I want to combine the tutorial 2 and tutorial 3 such that the laser scan maps of tutorial 3 is displayed on top of the static map which we have from tutorial 2. I have been trying to write a launch file for this setup but it does not work. I see the static map initially but as soon as I move the robot I only see the laser scans only and the static map disappears. Also, when I did the sudo apt-get install ros-hydro-nav2d-* all the packages just have a cmake and .xml file. How is this tutorial even working without a source code ?

skasperski, 2015-08-07:
If you start tutorial 2, you should already see the laser scans on top of the static map in RVIZ. So I assume you want to display both the static and the generated map simultaneously. It should be enough to set the generated map transparent, so the static one is still visible.

If you install the ubuntu-packages, only the compiled binaries and scripts are installed.If you want the source code, you need to clone it here from github.

ktiwari9, 2015-08-07:
Here is the screenshot of what I currently have. I want to ask why is the initial laser scan slightly offset from the actual walls (red lines and black lines are not aligned). Also as we can see that in the Stage scene the robot is slightly rotated but in my rviz scene its pointing dead east, I dont get why ?

Here is my final launch file : 

<launch>

	
	<param name="use_sim_time" value="true" />
	<rosparam file="$(find nav2d_tutorials)/param/ros.yaml"/>

	
	<node name="Stage" pkg="stage_ros" type="stageros" args="$(find auto_nav_sim)/world/auto_nav.world">
		<param name="base_watchdog_timeout" value="0" />
	</node>

	
	<node name="Operator" pkg="nav2d_operator" type="operator" >
		<remap from="scan" to="base_scan"/>
		<rosparam file="$(find nav2d_tutorials)/param/operator.yaml"/>
		<rosparam file="$(find auto_nav_sim)/param/costmap.yaml" ns="local_map" />
	</node>

	
	<node name="MapServer" pkg="map_server" type="map_server" args="$(find auto_nav_sim)/world/static_world.yaml" />

	
	<node name="SelfLocalizer" pkg="nav2d_localizer" type="localizer">
		<remap from="scan" to="base_scan"/>
		<param name="min_particles" type="int" value="5000"/>
		<param name="max_particles" type="int" value="20000"/>
	</node>

	
	<node name="Mapper" pkg="nav2d_karto" type="mapper">
		<remap from="scan" to="base_scan"/>
		<rosparam file="$(find nav2d_tutorials)/param/mapper.yaml"/>
	</node>
	
	
	<node name="Navigator" pkg="nav2d_navigator" type="navigator">
		<rosparam file="$(find nav2d_tutorials)/param/navigator.yaml"/>
	</node>

	<node name="GetMap" pkg="nav2d_navigator" type="get_map_client" />
	<node name="Explore" pkg="nav2d_navigator" type="explore_client" />
	<node name="SetGoal" pkg="nav2d_navigator" type="set_goal_client" />

	<include file="$(find using_markers)/launch/markers.launch" />

	
	
	<include file="$(find p2os_urdf)/launch/pioneer3at_urdf.launch" />

	
	<node name="RVIZ" pkg="rviz" type="rviz" args=" -d $(find auto_nav_sim)/param/auto_nav.rviz" />
	

</launch>

Please advise.
[image]

skasperski, 2015-08-07:
You should never have MapServer and Mapper at once. You should remove the Mapper completely if you just want to operate on a static map.

closed 2015-08-19 by skasperski