skasperski/navigation_2d #13: How to add obstacles to the scene ?

ktiwari9, 2015-08-07:
I want to add some obstacles to the scene in order to get the sensor reading from hokuyo laser scanner and then avoid obstacles while planning trajectory. How can I do this in Tutorial 3 ? Is it simply adding a Rviz Cylinder type marker ?
Please advise.

skasperski, 2015-08-07:
RVIZ is only for visulaization. If you want to add obstacles to the scene, you have to add it in Stage-Simulator. This can easily be done by adding a "model"-object to the world-file:

http://rtv.github.io/Stage/group__model.html

ktiwari9, 2015-08-07:
The problem is I want to simulate a mobile obstacle like a walking pedestrian. So, if I fix the pose of the obstacle as described in the file above I will not be able to make it move like a human who tend to walk on will and stop on will. Is there a way to simulate such mobile obstacles in rviz so that our robot can avoid it ?

skasperski, 2015-08-07:
Again: Nothing of this can be achieved in RVIZ. Anything that the robot should see has to be added to the simulation environment "Stage". So you have to think about how your pedestrians should be controlled: should they move on predefined paths or should they also do obstacle avoidance and path planning on their own?

ktiwari9, 2015-08-07:
I think pre-defined paths should be easy for a start since I have never used Stage-Sim before. So, how should I go about writing a pre-defined path for the pedestrians in order simulate their motion ?

ktiwari9, 2015-08-07:
Aah I see. So you mean initially I just load an obstacle at some random location. And then move it around in the stage file (manually) so that the robot will have to dynamically avoid it while it is going to the goal position that I selected by choosing the 2d Nav Goal?

skasperski, 2015-08-07:
You could just add them like other robots and write a small ROS-Node that controls your "pedestrians" via "cmd_vel" command topic. I think Stage can also define some simple behaviours for simulated agents, but I never used this before. You should check the Stage documentation for this.

skasperski, 2015-08-07:
Yes, that is exactly what I did to test the Operator's obstacle avoidance. I just dragged the object in front of the robot so it had to avoid it automatically.

ktiwari9, 2015-08-07:
That sounds fine for now. I think I will do it this way so that we can manually put the obstacle while the robot is in motion so that it has to dynamically avoid it. So should I define this "model" in the costmap.yaml file ?

ktiwari9, 2015-08-07:
The only problem is I need a way to visualize this obstacle ("pedestrian") in rviz. Is it possible that I can pick up my obstacle co-ordinates from stage sim and fetch it in my ros node ? Then I can simply put my rviz marker at that location I just fetched and it would look like I showed you in a previous screen shot but this time the robot will actually avoid it since it will see it an obstacle. So, it would solve my purpose.

skasperski, 2015-08-07:
I don't know whether this is possible. Better check with the Stage documentation first.

ktiwari9, 2015-08-07:
So, I changed the tutorial.world file to this :

include "p3at.inc"
include "floorplan.inc"

name                    "Navigation-2D Tutorial"
interval_sim            100
quit_time                 0
resolution                0.025
show_clock                0
show_clock_interval     100
threads                 2

# Load an environment bitmap
floorplan
( 
  name "tutorial"
  bitmap "autolab_fill.png"
  size [34 30 1.0]
  pose [0 0 0 0]
)

# Put a single robot into the world
pioneer3at
(
  name "robot"
  pose [0 -2 0 45]
  #pose [0 0 0 0]
)

model
(
    pose [ 0.0 0.0 0.0 0.0 ]
    size [ 0.1 0.1 0.1 ]
    origin [ 0.0 0.0 0.0 0.0 ]

    update_interval 100

    color "red"
    color_rgba [ 0.0 0.0 0.0 1.0 ]
    bitmap ""
    ctrl ""

    # determine how the model appears in various sensors
    fiducial_return 0
    fiducial_key 0
    obstacle_return 1
    ranger_return 1.0
    blob_return 1
    ranger_return 1.0
    gripper_return 0

    # GUI properties
  	gui_nose 0
   	gui_grid 0
    gui_outline 1
    gui_move 0 (1 if the model has no parents);

    boundary 0
    mass 10.0
    map_resolution 0.1
    say ""
    alwayson 0

   stack_children 1
 )

Am I missing something ?

ktiwari9, 2015-08-07:
I added an obstacle and you were right the green block is an obstacle and the robot does detect it. The problem is if I want to go somewhere beyond the obstacle the robot does not move. It detects an obstacle and stays there. Is this what we expected ?

[image]

skasperski, 2015-08-07:
The object is probably just a little to close. (The robot thinks it already crashed with it and won't move anymore.) Can you move it a little away from the robot?

closed 2015-08-19 by skasperski

aditi741997, 2020-09-18:
Is there any way to move the obstacle 'model' in stage, using some commandline tool? I wanted to move it along a straight line, back and forth [without using the GUI] for my experiments.