skasperski/navigation_2d #14: RVIZ crashes with tutorial launch files

skasperski, 2015-08-31:
Happens to me these days, can anyone confirm this issue?

closed 2017-02-02 by skasperski

scanf30, 2018-05-26:
how you can solve this?

i tried to run the tutorial 2 launch and, after the 1st spin mapping rviz clashes.

skasperski, 2018-05-28:
Which version of ROS are you using? Any error messages on RVIZ' command line?

bairuofei, 2023-03-07:
Has this problem been solved? I also encountered this problem when running tutorial launch files, and also when running another package developed upon it. The Rviz window will crash only after I zoom in or rotate in the display window, and works fine if I do not touch the display window in Rivz when running the simulation.

Systems:
Ubuntu18.04 laptop with ros version of melodic.

After I restart Rviz in another console and load the .rviz file, the Rviz can be recovered, but it will also crash when I zoom in the display window.

The error message in roslaunch console (the log file indicated in the error info is actually empty):

"[RVIZ-11] process has died [pid 32596, exit code -11, cmd /opt/ros/melodic/lib/rviz/rviz -d /home/ruofei/buffa_code/cpp/catkin_chinesePP_ws/src/chinese-postman-problem/src/ros_exploration_plugin/param/exploration.rviz __name:=RVIZ __log:=/home/ruofei/.ros/log/835e9c0c-bcd3-11ed-8044-00e04c68010d/RVIZ-11.log].
log file: /home/ruofei/.ros/log/835e9c0c-bcd3-11ed-8044-00e04c68010d/RVIZ-11*.log"

The error message in another console after running "rosrun rviz rviz" to restart rviz:
"Segmentation fault"

Do you have any idea how to fix it?

skasperski, 2023-03-07:
Hello. I haven't had this issue with any newer ROS version anymore, so I kinda forgot about it. 

Have you installed nav2d from source or the ros-packages?
Can you verify that it only happens with nav2d?

You could try not to load the configuration of rviz from the launch file. Maybe there is some incompatibility between the .rviz files from nav2d_tutorials and the current RVIZ. So instead launch RVIZ separately and add the visualizations one by one, to see which one causes the crash.

bairuofei, 2023-03-07:
Thank you for the prompt reply. 

I use nav2d from both source and ros-packages to load the Rviz, and the Rivz crashes in both cases when I call the service /StartExploration and adjust the display window. I think it only happens with nav2d because I did not find the same problem before when I ran other SLAM simulations.

Just now I tried to create a new Rviz file and added all the topics one by one, but still it crashed after I call the /StartExploration service and move the display window. Again, this seems will not happen if I do not touch the display window. And the crash is more likely to happen when I am zooming in the display window. The error info is the same as I posted above.

bairuofei, 2023-03-08:
After my test, I find the problem can be easily reproduced by only adding a Robotmodel topic and a Gridcells topic (/Operator/desired, or /Operator/route in nav2d package) to rviz. The rviz will be crashed instantly, with a "Segmentation fault" error.

Both topic individually works fine.