skasperski/navigation_2d #15: nav2d_karto: LocalizedRangeScan contains unexpected numbers of range readings

g-gemignani, 2015-11-11:
Hello,
I was able to run the third tutorial, mapping the example world provided in simulation.
At this point, I tried to create a map playing [a bag](https://drive.google.com/file/d/0Bx1My6gkqxtyRXluWndoRXE5REU/view?usp=sharing) recorded with our robot (the bag seems ok since I was able to use it both with hector_slam and gmapping). The launch file adopted is the following:
```xml
<launch>

	<param name="use_sim_time" value="true" />
	<rosparam file="ros.yaml"/>
	
	<!-- Start Mapper to genreate map from laser scans -->
	<node name="Mapper" pkg="nav2d_karto" type="mapper">
		<remap from="scan" to="scan_front"/>
		<rosparam file="mapper.yaml"/>
	</node>

	<!-- RVIZ to view the visualization -->
	<node name="RVIZ" pkg="rviz" type="rviz" args=" -d $(find nav2d_tutorials)/param/tutorial3.rviz" />

</launch>

```
where ros.yaml is:
```
###########################################################
# Defines topics services and frames for all modules

### TF frames #############################################
laser_frame: base_front_laser_link
robot_frame: base_link
odometry_frame: odom
offset_frame: offset
map_frame: map

### ROS topics ############################################
map_topic: map
laser_topic: scan_front

### ROS services ##########################################
map_service: static_map
```
and mapper.yaml is:
```
### Mapper ################################################

grid_resolution: 0.1
range_threshold: 10.0
map_update_rate: 5
publish_pose_graph: true
max_covariance: 0.01
transform_publish_period: 0.1
min_map_size: 20

### Localizer #############################################

min_particles: 2500
max_particles: 10000

### Karto #################################################

# For a full list of available parameters and their
# corresponding default values, see OpenKarto/OpenMapper.h

MinimumTravelDistance: 1.0
MinimumTravelHeading: 0.52
LoopSearchMaximumDistance: 10.0

###########################################################
```
However, when I run the launch file, I repeatedly get the following error:
```
...
[/Mapper ERROR 1447255247.422253465, 1446138392.065241589]: LaserRangeFinder::Validate() - LocalizedRangeScan contains 450 range readings, expected 451
[/Mapper ERROR 1447255247.473642245, 1446138392.166068413]: LaserRangeFinder::Validate() - LocalizedRangeScan contains 450 range readings, expected 451
[/Mapper ERROR 1447255247.523607012, 1446138392.266683794]: LaserRangeFinder::Validate() - LocalizedRangeScan contains 450 range readings, expected 451
[/Mapper ERROR 1447255247.573577659, 1446138392.367428587]: LaserRangeFinder::Validate() - LocalizedRangeScan contains 450 range readings, expected 451
[/Mapper ERROR 1447255247.573836632, 1446138392.367428587]: LaserRangeFinder::Validate() - LocalizedRangeScan contains 450 range readings, expected 451
[/Mapper ERROR 1447255247.624776753, 1446138392.468281149]: LaserRangeFinder::Validate() - LocalizedRangeScan contains 450 range readings, expected 451
[/Mapper ERROR 1447255247.674386008, 1446138392.569080442]: LaserRangeFinder::Validate() - LocalizedRangeScan contains 450 range readings, expected 451
[/Mapper ERROR 1447255247.724970486, 1446138392.669702679]: LaserRangeFinder::Validate() - LocalizedRangeScan contains 450 range readings, expected 451
[/Mapper ERROR 1447255247.775455585, 1446138392.770533571]: LaserRangeFinder::Validate() - LocalizedRangeScan contains 450 range readings, expected 451
[/Mapper ERROR 1447255247.775564465, 1446138392.770533571]: LaserRangeFinder::Validate() - LocalizedRangeScan contains 450 range readings, expected 451
...
```
Am I missing something or there is a problem with the code?
Thanks!

skasperski, 2015-11-11:
The number of scans should be calculated from min_angle, max_angle and angle_increment as stated in the LaserScan-Message from ROS. Have you checked whether these values are correct? This might even be a fence-post-error.

g-gemignani, 2015-11-11:
Thank you for the quick reply.
I am not sure if I understood correctly. I checked if the messages sent from the bag are correct. 
These messages have the following parameters (I got them by echoing the topic on which they are published):
angle_min: -1.96349537373
angle_max: 1.96349537373
angle_increment: 0.00872664619237
which give me: 449,999995519 messages. I also checked in those messages if there were too many range readings by I still got the correct number (450)...
Thanks again!

skasperski, 2015-11-12:
If you have 450 segments [(max-min)/increment] in your scan-range, that means you should have 451 scans. (There is a scan at each end of the interval, see here: https://en.wikipedia.org/wiki/Off-by-one_error#Fencepost_error)

So your scans should indeed contain 451 scans.

g-gemignani, 2015-11-12:
Thank you for your replay.
Then, I guess that the pull request that I created is not correct. I now suspect that there might be something wrong in the laser drivers that I am using. (maybe they are truncating [(max-min)/increment]). In any case, thank you so much for your help, I am now able to use the mapper.
Cheers,
Guglielmo

closed 2015-12-01 by skasperski