skasperski/navigation_2d #17: Issues setting up with turtlebot 2

pgigioli, 2016-03-21:
I've spent a lot of timing trying to customize the tutorial3.launch file for a turtlebot2 so that I can have autonomous gmapping and exploration but I am running into a lot of issues.  Firstly, I can't seem to get the mapper node to publish a map to the operator costmap or to anything else for that matter.  I'm running the turtlebot_bringup/minimal.launch and a custom version of the tutorial3.launch.  RVIZ is run on a separate computer.

Here is my rqt_graph:

[image]

Here is the tutorial3.launch:

`   <launch>

        
        <param name="use_sim_time" value="false" />
        <rosparam file="$(find nav2d_tutorials)/param/ros.yaml"/>

        
        <node name="Operator" pkg="nav2d_operator" type="operator" >
                
                <remap from="cmd_vel" to="cmd_vel_mux/input/teleop"/>
                <rosparam file="$(find nav2d_tutorials)/param/operator.yaml"/>
                <rosparam file="$(find nav2d_tutorials)/param/costmap.yaml" ns="local_map" />
        </node>

        
        <node name="Mapper" pkg="nav2d_karto" type="mapper">
                
                <rosparam file="$(find nav2d_tutorials)/param/mapper.yaml"/>
        </node>

        
        <node name="Navigator" pkg="nav2d_navigator" type="navigator">
                <rosparam file="$(find nav2d_tutorials)/param/navigator.yaml"/>
        </node>

        <node name="GetMap" pkg="nav2d_navigator" type="get_map_client" />
        <node name="Explore" pkg="nav2d_navigator" type="explore_client" />
        <node name="SetGoal" pkg="nav2d_navigator" type="set_goal_client" >
                <remap from="goal" to="move_base_simple/goal"/>
   </launch>
`

Can you help me with this setup?  I know this has been done before.  Let me know if you need to see any param files.

skasperski, 2016-03-29:
Looking at your ros-graph, there doesn't seem to be any laser scans going into the Operator or the Mapper. This way neither obstacle avoidance nor mapping can work. So check to which topic your scans are published and remap Operator and Mapper inputs accordingly.

pgigioli, 2016-03-29:
Please see [this](http://answers.ros.org/question/229994/nav2d-startexploration-2-fails-turtlebot2/) post for my latest update on this issue.  I have been able to get a map published and have gotten as far as getting /StartMapping to work.  Right now, /StartExploration runs without any errors but the robot doesn't do anything.

closed 2017-02-02 by skasperski