skasperski/navigation_2d #19: navigation_tutorial tutorial3.launch crashes mapper

manuel-delverme, 2016-05-15:
Running tutorial3.launch results in a crash  
```
[Mapper-4] process has died [pid 11039, exit code -11, cmd /opt/ros/jade/lib/nav2d_karto/mapper scan:=base_scan __name:=Mapper __log:=/home/noflip/.ros/log/b6b6ae70-1acb-11e6-b2f9-1c659dee480d/Mapper-4.log].
log file: /home/noflip/.ros/log/b6b6ae70-1acb-11e6-b2f9-1c659dee480d/Mapper-4*.log
```

position of the robot:
[image]

[image]
(i wouldn't trust rviz here, i think it didn't refresh the position since i was in another worskpace)

```
[terminal:1]
➜  launch git:(master) ✗ roslaunch nav2d_tutorials tutorial3.launch

[terminal:2]
➜  launch git:(master) ✗ rosservice call /StartMapping 3
response: 0
➜  launch git:(master) ✗ rosservice call /StartExploration 2
response: 0

[terminal:1]
[Mapper-4] process has died [pid 13336, exit code -11, cmd /opt/ros/jade/lib/nav2d_karto/mapper scan:=base_scan __name:=Mapper __log:=/home/$USER/.ros/log/da07a914-1acc-11e6-b2f9-1c659dee480d/Mapper-4.log].
log file: /home/$USER/.ros/log/da07a914-1acc-11e6-b2f9-1c659dee480d/Mapper-4*.log

```
log file is empty

same exception happens, in another location after adding respawn="true" to mapper in the launch file

manuel-delverme, 2016-05-15:
[image]
same thing happened here, maybe it's not the specific position

skasperski, 2016-05-17:
Which version of ROS are you using? Have you installed binaries or build it from source? Does the crash happen immediately after the service call? 

Also check rqt_console for any hints just before the crash happens.

manuel-delverme, 2016-05-18:
Jade binaries from Ubuntu repository
The crash happens in choke points like the ones in the screenshots 
I'll try to post rqt console but I'll take me a while

skasperski, 2017-02-02:
Sorry for late reply. I have looked into this a bit and it seems to happen sometimes when a new map is requested. (New requests are likely to happen at narrow points.) It seems to be caused by some (changed?) behavior of libTBB inside of OpenKarto, but can't say anything for sure yet. It happens on a source-build with ros-kinetic as well.

skasperski, 2017-03-07:
Solved in version 0.3.1.

closed 2017-03-07 by skasperski