skasperski/navigation_2d #20: robot travels in circles

mariiiiiia, 2016-05-19:
Hi, I wanted to ask why the robot travels almost in circles, defining a very long path in order to reach the target, although the target is near to the robot. 
Is there a parameter to fix it?

Thanks

skasperski, 2016-05-19:
Hi, it's hard to judge from your description, can you post a picture with the map and the generated path? Is the goal probably near an obstacle?

mariiiiiia, 2016-05-19:
Here is an example of the path travelled and the target.
[image]

I think that the problem takes place wherever I put the target.

skasperski, 2016-05-20:
Where was the target point? Please show the planned path (topic "plan" from navigator node).

mariiiiiia, 2016-05-20:
In the previous screenshot the red arrow was the target and the red lines were the path that the robot traveled.
When i select "plan" from navigator node I only see the green curve you see in the screenshot. It's a 'permanent' path that is always being updated until robot reaches the target. And leads the robot in many random curves around target. (But it reaches it's target, so I think it's not the navigation_goal_distance parameter the problem).

This time I noticed something useful:  the 'map' frame axes visualized in rviz is often changing coordinates. Shouldn't it be stable the whole time? In the screenshot you can see my tf tree.

[image]

skasperski, 2016-05-20:
In your image, the red arrow points directly to an obstacle. The robot can only traverse the grey and dark blue area, so it cannot reach this target. It looks like the global map and local costmap are not aligned at all. I think you should solve this problem first, because navigation depends on this. 

Is your localization right? Is the costmap correctly updated? I would suggest to run the tutorial3 from nav2d_tutorials on your pc to get a feel of how this all should work together. I think this will help you to see what is going wrong on your real robot. Unfortunately I cannot see it just from the screenshots.

The "min_replanning_period" is not so critical, I would probably set it to 1.0 second or so, but this is really secondary optimization.

mariiiiiia, 2016-05-20:
The target goal is in the origin of the arrow, not at the end. So the robot theoretically could reach the target. 

Also, I have done the tutorials in simulation and worked fine. So I'm gonna argue with you about the costmap. I think they are aligned well. The only difference is that the local costmap is a little smaller than the global map, that's why you don't see the whole of it.

skasperski, 2016-05-20:
Ah, ok. But still the target is very near to an obstacle. Have you tried to move it into the open area of your room?

closed 2017-02-02 by skasperski