skasperski/navigation_2d #23: How to create a fixed origin map when use

jiangxinyu, 2016-08-12:
I want to create a fix-origin position map when use the exploration. However when the origin position of the map changed everytime when I restart the node.How should do?

skasperski, 2016-08-16:
I don't think that I really understand your question. When you start the mapping, the robot will always start at (0,0) in world coordinates. So if you want to build a similar map again, you have to start the robot in the same place where you started before.

The "origin" value of the OccupancyGrid however will change during the mapping process. (Not only when reatarting the mapping) This is because the map grows over time, but the map origin always marks the lower left edge of the map. But you can always use this origin to translate from world coordinates to map coordinates.

jiangxinyu, 2016-08-18:
OK, I have got it， thank you very much

closed 2016-08-18 by skasperski