skasperski/navigation_2d #25: Tutorial 3 mapper error

alhouty808, 2016-12-07:
Hello, 
I would appreciate your help on this 

http://answers.ros.org/question/249500/nav2d-toturial-3/

Thanks,

alhouty808, 2016-12-08:
I provided the log file as requested.

skasperski, 2016-12-08:
I tested everything on a similar system, with Indigo on 14.04, but I can't reproduce the error.

There are some strange warnings in your Operator-Log. I don't know if it is related, but it might be a hint. Can you check the transform tree (rosrun tf view_frames), if anything is weird there?

When do this errors occur? After you called one of those services? If nothing else helps, you might try to build nav2d from source, just to see if any problems with your systems pop up.

alhouty808, 2016-12-08:
I will try to build the package from source. 

Thank you for you response.

alhouty808, 2016-12-09:
I just built the package from the source, and i faced 2 errors. I updated system's packages and built the package one more time, the error still pops up. 

[screenshot](https://s30.postimg.org/wjr3ca09d/oie_SCMc_LB6_Ooe9_S.pngl)

skasperski, 2016-12-09:
It says it can't find Costmap2D::mutex_t, but it should be declared in costmap_2d.h#294. Please do a "roscd costmap_2d" and see if it brings you to /opt/ros/indigo/share/costmap_2d. The file should then be located in /opt/ros/indigo/include/costmap_2d/costmap_2d.h

alhouty808, 2016-12-09:
i did the command and it brings me to  "/opt/ros/indigo/share/costmap_2d"

i open the file at "/opt/ros/indigo/include/costmap_2d/costmap_2d.h" 

and i guess the file has been modified here is a [screenshot](https://s17.postimg.org/i1k5cgh9r/oie_uva4du_Cb2n_Pg.png)

skasperski, 2016-12-09:
Ok, this is getting really strange. Can you please check again your version of this ubuntu-package:
ros-indigo-costmap-2d

Mine is:
1.12.13-0trusty-20161026-140913-0700

alhouty808, 2016-12-09:
I re-installed the ROS. Now everything works perfectly. 

Thanks,

closed 2016-12-09 by alhouty808