skasperski/navigation_2d #26: multi robot exploration

COE-koko, 2016-12-09:
dear Dr. Kasperski,

I am a Computer Engineering student at KFUPM. My senior project involves multi robot exploration and mapping. I just tested your package and did the tutorials. Now, i would like to deploy multiple robots in unknown environment to produce a map of that environment. so i created a launch file to simulate the process. but unfortunately, the environment don't get explored completely.

this is my launch file
`<launch>

	
	<param name="use_sim_time" value="true" />

	
	<node name="Stage" pkg="stage_ros" type="stageros" args="$(find nav2d_tutorials)/world/tutorial4.world">
		<param name="base_watchdog_timeout" value="0" />
	</node>

	<node name="R0_MapAlign" pkg="tf" type="static_transform_publisher" args="0 0 0 0 0 0 /map /robot_0/map 100"/>
	<node name="R1_MapAlign" pkg="tf" type="static_transform_publisher" args="40 0 0 0 0 0 /map /robot_1/map 100"/>

	<group ns="robot_0">
		<param name="robot_id" value="1" />
		<param name="tf_prefix" type="string" value="robot_0"/>
		<rosparam file="$(find nav2d_tutorials)/param/ros.yaml"/>

		<node name="Mapper" pkg="nav2d_karto" type="mapper">
			<remap from="scan" to="base_scan"/>
			<remap from="karto_in" to="/shared_scans_r2"/>
			<remap from="karto_out" to="/shared_scans_r1"/>
			<rosparam file="$(find nav2d_tutorials)/param/mapper.yaml"/>
		</node>

		<node name="Operator" pkg="nav2d_operator" type="operator" >
		<remap from="scan" to="base_scan"/>
		<rosparam file="$(find nav2d_tutorials)/param/operator.yaml"/>
		<rosparam file="$(find nav2d_tutorials)/param/costmap.yaml" ns="local_map" />
	</node>

	
		<node name="Navigator" pkg="nav2d_navigator" type="navigator">
			<rosparam file="$(find nav2d_tutorials)/param/navigator.yaml"/>
		</node>

		<node name="Localize" pkg="nav2d_navigator" type="localize_client" />
		<node name="Explore" pkg="nav2d_navigator" type="explore_client" />
		<node name="GetMap" pkg="nav2d_navigator" type="get_map_client" />

	</group>

	<group ns="robot_1">
		<param name="robot_id" value="2" />
		<param name="tf_prefix" type="string" value="robot_1"/>
		<rosparam file="$(find nav2d_tutorials)/param/ros.yaml"/>

		
		<node name="Operator" pkg="nav2d_operator" type="operator" >
			<remap from="scan" to="base_scan"/>
			<rosparam file="$(find nav2d_tutorials)/param/operator.yaml"/>
			<rosparam file="$(find nav2d_tutorials)/param/costmap.yaml" ns="local_map" />
		</node>

		
		<node name="Navigator" pkg="nav2d_navigator" type="navigator">
			<rosparam file="$(find nav2d_tutorials)/param/navigator.yaml"/>
		</node>

		<node name="Localize" pkg="nav2d_navigator" type="localize_client" />
		<node name="Explore" pkg="nav2d_navigator" type="explore_client" />
		<node name="GetMap" pkg="nav2d_navigator" type="get_map_client" />
		<node name="Mapper" pkg="nav2d_karto" type="mapper">
			<remap from="scan" to="base_scan"/>
			<remap from="karto_in" to="/shared_scans_r1"/>
			<remap from="karto_out" to="/shared_scans_r2"/>
			<rosparam file="$(find nav2d_tutorials)/param/mapper.yaml"/>
		</node>

	</group>

	
	<node name="RVIZ" pkg="rviz" type="rviz" args=" -d $(find nav2d_tutorials)/param/tutorial4.rviz" />

</launch>
`
[RViz](https://s28.postimg.org/3lgmw7b2l/Screenshot_from_2016_12_09_15_21_30.png)
[Stage](https://s28.postimg.org/bs8mns159/Screenshot_from_2016_12_09_15_21_44.png)

skasperski, 2016-12-09:
These usually happens because the second robot was not correctly localized. The first robot has to explore enough around the start point and the second has to move in this explored area before it starts exploring itself. Check the localization particles to see if they converge to the correct pose of the second robot.

alhouty808, 2016-12-09:
Thank you for your comment. I will check i get back to you