skasperski/navigation_2d #32: Missed desired rate of 5.00Hz

alhouty808, 2017-01-01:
Hello, 

when i used exploration strategy "MultiWavfrontPlanner" for multi robot, I get this message" Missed desired rate of 5.00Hz! Loop actually took 0.5000 seconds!" and robot 2 will turn 360 for every meter. 

hope you can help

skasperski, 2017-01-02:
When a node cannot keep its desired rate, it usually means that your system load is too high, which can cause all kinds of bad side effects.

The behaviour of the second robot is actually the intended one. It will repeat the localize action (move 2m and turn 360°) indefinitely until it is localized. So you should find out why localization is not working. Most likely it is because your map from robot 1 is not large enough.

alhouty808, 2017-01-02:
I'm running all nodes on my machine(core i5 & 16GB ram) with stage simulator. should it be enough ?

skasperski, 2017-01-02:
Should be enough, but you can check the system monitor application to see if any core runs on 100%.

alhouty808, 2017-01-02:
do the robots need any further setup (start exploration in the same direction ?).

skasperski, 2017-01-02:
No, it is only important to explore a large area around the start pose before starting the second robot, and the start area should be reasonably feature rich. The initialization is definitely the most critical phase of the multi-robot exploration.

alhouty808, 2017-01-02:
there is a core that peaks 100%, after 2 min. what would you suggest, since i face similar issue with "MinPosPlanner".

skasperski, 2017-01-02:
Find out which node uses so much CPU. If it's the planner, maybe reduce map resolution.

alhouty808, 2017-01-02:
where is the map resolution parameter? 
is it in the costmap.yaml(resolution: 0.05) or mapper.yaml(grid_resolution: 0.1)

skasperski, 2017-01-02:
In the "mapper.yaml" file, "costmap.yaml" is for the operator's local costmap. But 0.1 is already quite small. Which node uses so much CPU time?

alhouty808, 2017-01-02:
mapper of robot_1

skasperski, 2017-01-02:
Have you changed MinimumTravelDistance and MinimumTravelHeading? If these are too low, they can cause the mapper to require a lot of CPU time.

alhouty808, 2017-01-02:
MinimumTravelDistance: 1.0
MinimumTravelHeading: 0.52
LoopSearchMaximumDistance: 10.0
may i ask what they represent ?

skasperski, 2017-01-02:
Please refer to [OpenMapper.h](https://github.com/skasperski/navigation_2d/blob/master/nav2d_karto/OpenKarto/source/OpenKarto/OpenMapper.h#L1389)

closed 2018-05-23 by skasperski