skasperski/navigation_2d #33: mapper not generating expected occupancy grid map

nitin5, 2017-06-01:
Hi,
 
 I have been running a few experiments with 2D lidar data. I found that mapper do not generates complete occupancy grid initially. I am attaching two pictures. Please see that a lot of area is left as unknown in the map although, laser beams has passed through them.
 [image]
 This picture shows starting position of the robot. Size of a cell in rviz display is 0.25x0.25 m-sq while size of a cell in Stage grid is 1m-sq.  Laser range in simulation is 10meters. 

Further, Robot starts moving backward. Next picture is output after some movement backward. 
 [image]
 
 
> I have been using following configuration data
> ### Mapper ################################################
> robot_id: 1
> grid_resolution: 0.25
> range_threshold: 10.0 #Maximum range of laser sensor
> map_update_rate: 1
> publish_pose_graph: true
> max_covariance: 0.01 # Threshold to accept result of particle filter
> transform_publish_period: 0.1
> min_map_size: 20 #minimum number of nodes in map for localization
> #laser_topic: base_scan
> #map_frame: /map
> #map_topic: map
> #laser_frame: base_laser_link
> #robot_frame: base_link
> #odometry_frame: odom1
> #offset_frame: offset
> ### Localizer #############################################
> 
> min_particles: 2500
> max_particles: 10000
> 
> ### Karto #################################################
> 
> MinimumTravelDistance: 1.0
> MinimumTravelHeading: .52
> LoopSearchMaximumDistance: 5.0
> ScanBufferMaximumScanDistance: 20.0
> ScanBufferSize: 8
> 
> 
> # For a full list of available parameters and their
> # corresponding default values, see OpenKarto/OpenMapper.h
> #UseScanMatching: true
> #UseScanBarycenter: true
> #MinimumTravelDistance: 1.0
> #MinimumTravelHeading: 0.52
> #ScanBufferSize: 70
> #ScanBufferMaximumScanDistance: 20
> #UseResponseExpansion: false
> #    Variance of penalty for deviating from odometry when scan-matching.  Use lower values if the robot's odometer is more accurate.
> #DistanceVariancePenalty: 0.6
> #MinimumDistancePenalty: 0.5
> #AngleVariancePenalty: 0.1225 # variance of 20 x 20 degree sq in radians
> #MinimumAnglePenalty: 0.9
> #LinkMatchMinimumResponseFine: 0.6
> #LinkScanMaximumDistance: 5.0
> #CorrelationSearchSpaceDimension: 0.3 #Seach for correlation in area of 30x30 cm-square
> #CorrelationSearchSpaceResolution: 0.1
> #CorrelationSearchSpaceSmearDeviation: 0.03
> #CoarseSearchAngleOffset: 0.34906585 #20 degrees
> #FineSearchAngleOffset: 0.00349 #0.2degrees
> #CoarseAngleResolution: 0.0349 #2 degrees
> #LoopSearchSpaceDimension: 8 #search area of size 8x8 meter square
> #LoopSearchSpaceResolution: 0.05
> #LoopSearchSpaceSmearDeviation: 0.03
> #LoopSearchMaximumDistance: 10.0
> #When the loop closure detection finds a candidate it must be part of a large set of linked scans. If the chain of scans is less than this value, we do not attempt to close the loop.
> #LoopMatchMinimumChainSize: 10 
> #LoopMatchMaximumVarianceCoarse: 0.16 #The co-variance values for a possible loop closure have to be less than this value  to consider a viable solution
> #LoopMatchMinimumResponseCoarse: 0.7 #if probability value is larger than this, initiate coarser level search
> #LoopMatchMinimumResponseFine: 0.7 #if probability value is larger than this, initiate finer level search
> 
> ###########################################################

Any help in how can i refine output of mapping will be appreciated.

skasperski, 2017-06-01:
Is it intentional to use such a low resolution for the map? I would opt for a much higher one for indoor mapping (e.g. 0.05). And try to increase the mapping range_threshold as well. Maybe the rest of your points is out of this range.

nitin5, 2017-06-01:
Yes, low resolution use was intentional to enable faster testing and debug.
I agree 5cm resolution is better choice for experiments on robots.
I realized that output is influenced by the max range of sensor. If I set
ranger's max range value to higher values (10 or 20 meter), the output from
nav2d_karto is much better. otherwise, there is tendency to leave out those
rays which do not hit any obstacles i.e. range reading is max range value.
This behaviour I have also observed in costmap2d_ros. Costmap2d_ros do not
consider those laser beams for which reading is maximum value.
Rest of the points are not out of range of laser sensor beams although,
still grid cells are not clear.

skasperski, 2017-06-02:
I am not sure I fully understand the problem. The pictures look okay to me, given the shown laser scans and the low map resolution. Which cells should be cleared, that are not?

nitin5, 2017-06-16:
I realised that problem can be avoided using a finer resolution or long range sensor, at least in simulations.

closed 2017-06-16 by nitin5