skasperski/navigation_2d #41: The branch with_g2o can't be compiled successfully

drscopus, 2017-11-07:
The compilation errors are following:
`/home/scopus/scopus_planning_ws/src/karto_g2o/nav2d_karto/src/G2oSolver.cpp: In constructor ‘G2oSolver::G2oSolver()’:
/home/scopus/scopus_planning_ws/src/karto_g2o/nav2d_karto/src/G2oSolver.cpp:27:65: error: no matching function for call to ‘g2o::BlockSolver<g2o::BlockSolverTraits<-1, -1> >::BlockSolver(SlamLinearSolver*&)’
  SlamBlockSolver* blockSolver = new SlamBlockSolver(linearSolver);
                                                                 ^
/home/scopus/scopus_planning_ws/src/karto_g2o/nav2d_karto/src/G2oSolver.cpp:27:65: note: candidate is:
In file included from /usr/local/include/g2o/core/block_solver.h:199:0,
                 from /home/scopus/scopus_planning_ws/src/karto_g2o/nav2d_karto/src/G2oSolver.cpp:3:
/usr/local/include/g2o/core/block_solver.hpp:40:1: note: g2o::BlockSolver<Traits>::BlockSolver(std::unique_ptr<typename Traits::LinearSolverType>) [with Traits = g2o::BlockSolverTraits<-1, -1>; typename Traits::LinearSolverType = g2o::LinearSolver<Eigen::Matrix<double, -1, -1> >]
 BlockSolver<Traits>::BlockSolver(std::unique_ptr<LinearSolverType> linearSolver)
 ^
/usr/local/include/g2o/core/block_solver.hpp:40:1: note:   no known conversion for argument 1 from ‘SlamLinearSolver* {aka g2o::LinearSolverCholmod<Eigen::Matrix<double, -1, -1> >*}’ to ‘std::unique_ptr<g2o::LinearSolver<Eigen::Matrix<double, -1, -1> >, std::default_delete<g2o::LinearSolver<Eigen::Matrix<double, -1, -1> > > >’
/home/scopus/scopus_planning_ws/src/karto_g2o/nav2d_karto/src/G2oSolver.cpp:28:79: error: no matching function for call to ‘g2o::OptimizationAlgorithmGaussNewton::OptimizationAlgorithmGaussNewton(SlamBlockSolver*&)’
  mOptimizer.setAlgorithm(new g2o::OptimizationAlgorithmGaussNewton(blockSolver));
                                                                               ^
/home/scopus/scopus_planning_ws/src/karto_g2o/nav2d_karto/src/G2oSolver.cpp:28:79: note: candidate is:
In file included from /home/scopus/scopus_planning_ws/src/karto_g2o/nav2d_karto/src/G2oSolver.cpp:6:0:
/usr/local/include/g2o/core/optimization_algorithm_gauss_newton.h:46:16: note: g2o::OptimizationAlgorithmGaussNewton::OptimizationAlgorithmGaussNewton(std::unique_ptr<g2o::Solver>)
       explicit OptimizationAlgorithmGaussNewton(std::unique_ptr<Solver> solver);
                ^
/usr/local/include/g2o/core/optimization_algorithm_gauss_newton.h:46:16: note:   no known conversion for argument 1 from ‘SlamBlockSolver* {aka g2o::BlockSolver<g2o::BlockSolverTraits<-1, -1> >*}’ to ‘std::unique_ptr<g2o::Solver>’
make[2]: *** [karto_g2o/nav2d_karto/CMakeFiles/mapper.dir/src/G2oSolver.cpp.o] Error 1
make[1]: *** [karto_g2o/nav2d_karto/CMakeFiles/mapper.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
`

skasperski, 2017-11-07:
Have you tried building g2o from source? I remember there was an issue with the version that ships with ROS.

drscopus, 2017-11-08:
I  downloaded g2o from [here](https://github.com/RainerKuemmerle/g2o) , built and installed it. So I think g2o is ok. Thank you!

tyuownu, 2017-11-09:
I think you should install a special version of g2o, maybe the newest not compatible with old versions.

closed 2018-05-23 by skasperski