skasperski/navigation_2d #43: No map generated using turtlebot3

magnotanto, 2018-04-11:
Hi!
I'm trying to use the nav2d package for Turtlebot3's simulation using Gazebo instead of Stage. The problem is that when i launch the simulation(with third tutorial's routine), no map is generated and because of that, the exploration and mapping(of course) don't work. I've tried to control topics and as one can guess, no messages are published on /map topic. The fact is that i haven't touched anything, except adding Turtlebot3 model and Gazebo to launch file and remapped the laserscan topics. Checking rqt_graph showed the absence of /Operator/desired, /Operator/route and /Navigator/plan topics, so i think i'm missing some remapping. Is that the case? Could i have any help to understand what i'm missing?
 `
<launch>

	
	<rosparam file="$(find nav2d_tutorials)/param/ros.yaml"/>

	
	<arg name="model" default="$(env TURTLEBOT3_MODEL)" doc="model type [burger, waffle, waffle_pi]"/>
    <arg name="x_pos" default="-3.0"/>
    <arg name="y_pos" default="1.0"/>
    <arg name="z_pos" default="0.0"/>

    <include file="$(find gazebo_ros)/launch/empty_world.launch">
      <arg name="world_name" value="$(find turtlebot3_gazebo)/worlds/turtlebot3_house.world"/>
      <arg name="paused" value="false"/>
      <arg name="use_sim_time" value="true"/>
      <arg name="gui" value="true"/>
      <arg name="headless" value="false"/>
      <arg name="debug" value="false"/>
    </include>

	<param name="robot_description" command="$(find xacro)/xacro.py $(find turtlebot3_description)/urdf/turtlebot3_$(arg model).urdf.xacro"/>

    <node name="spawn_urdf" pkg="gazebo_ros" type="spawn_model" args="-urdf -model turtlebot3 -x $(arg x_pos) -y $(arg y_pos) -z $(arg z_pos) -param robot_description" >
    </node>

	
	<node name="Operator" pkg="nav2d_operator" type="operator" >
		<remap from="base_scan" to="scan"/>
		<rosparam file="$(find nav2d_tutorials)/param/operator.yaml"/>
		<rosparam file="$(find nav2d_tutorials)/param/costmap.yaml" ns="local_map" />
	</node>

	
	<node name="Mapper" pkg="nav2d_karto" type="mapper">
		<remap from="base_scan" to="scan"/>
		<rosparam file="$(find nav2d_tutorials)/param/mapper.yaml"/>
	</node>

	
	<node name="Navigator" pkg="nav2d_navigator" type="navigator">
		<rosparam file="$(find nav2d_tutorials)/param/navigator.yaml"/>
	</node>

	<node name="GetMap" pkg="nav2d_navigator" type="get_map_client" />
	<node name="Explore" pkg="nav2d_navigator" type="explore_client" />
	<node name="SetGoal" pkg="nav2d_navigator" type="set_goal_client" />

	
	<node name="Joystick" pkg="joy" type="joy_node" />
	<node name="Remote" pkg="nav2d_remote" type="remote_joy" />

	
	<include file="$(find turtlebot3_gazebo)/launch/turtlebot3_gazebo_rviz.launch">
		<arg name="model" value="$(arg model)"/>
    </include>
</launch>

`
This is my launch file.
[image]
And this is a screenshot of RViz.
Thanks for any help!!

skasperski, 2018-04-12:
I suggest to follow the flow of the data through the whole system.
- Are there laser scans published?
- Is the local map build?
- Are the transformations available? (Check with RVIZ and view_frames)

Also check rqt_console if there are any warnings or errors.

magnotanto, 2018-04-12:
The scans are published and trasformations are avaibles. Checking rqt_console gives warnings, it says "Failed to compute odometry pose, skipping scan ("base_laser_link" passed to lookupTransform argument source_frame does not exist. )"

magnotanto, 2018-04-12:
After i've answered you, i've googled the warning and i've found the solution in another issue you answered time ago(  [https://answers.ros.org/question/219046/nav2d-navigator-failing-to-create-path/](url) ). I've modified the laser_frame parameter and everything works well; i'm new and  i didn't know how useful rqt_console is. Thank you for your answer and patience.

closed 2018-04-12 by magnotanto