skasperski/navigation_2d #45: Output Command

CapitalZe, 2018-05-14:
I really enjoy Nav2d and have the simulations/tutorials up and running perfectly.
I even have it so that when I control the simulated robot in Rviz/Stage, my physical robot also reflects the control commands from the Joystick; such as turning angle and velocity.

The Nav2d, however, does not directly control my physical robot. If I set it to autonomous (for example in tutorial3.launch) then the robot in rviz drives around but my physical one does nothing, not until I use the joystick.

I believe my issue is with the output command messages. I am using the same robot from jetsonhacks (RaceCarJ).
How can I change the output command to ackermann_msgs/AckermannDriveStamped? 

Also, to control a physical robot would the output to the robot come from the Navigator Pkg or the Operator Pkg?

Thanks very much for your time!

skasperski, 2018-05-15:
Hello,

thanks for using nav2d! Please check the connections of your joystick. From your description I assume, that your joystick is connected directly to the physical robot. This way the joystick overrides the commands from the Navigator and the robot cannot move by itself. You should use rqt_graph to carefully check all connections. Only the Operator should write to the topic "/cmd_vel".

The connection of the joystick should be:
Joystick --(joy)--> Remote --(cmd)--> Operator --(cmd_vel)--> YourRobot

About the Navigator: The Navigator (for global planning) writes it output to the Operator (for local obstacle avoidance) and again, only the Operator writes to the physical robot. So for autonomous movement the connections are:
Navigator --(cmd)--> Operator --(cmd_vel)--> YourRobot

CapitalZe, 2018-05-15:
Hello,

Thanks very much for that input.

I was a bit confused by cmd and cmd_vel, your explanation makes perfect sense.

You also guessed completely right, I had two joy packages running, one directly to my robot and the other was the nav2d_remote package.

I will work on this today and let you know if I have any success.

Thanks for Nav2d, I think its a great resource to help teach about ROS, different forms of autonomy and to have fun overall.

CapitalZe, 2018-05-17:
I got it to work.

Your diagnosis and recommendation was correct.

I needed to add an intermediary node that converts twist to ackermann msgs, but that was not too difficult.

If anyone else is interested, I can post the script for the node here, which also creates a new topic to publish the converted messages across.

closed 2018-05-21 by CapitalZe