skasperski/navigation_2d #46: Button Mapping

CapitalZe, 2018-05-17:
I am using a logitech F170 controller.

The button numbers as designated in the remote_joy.py file do not seem to correspond to my logitech.
I can't find any other calibration file in the nav2d system.

Furthermore, forward and reverse work when controlling both the simulation and the real robot, but steering angle only works on the simulation.

Steering angle does work, however, on the physical robot when in autonomous mode.

Any ideas?

skasperski, 2018-05-22:
I use the F710 as well. Maybe you have to change the mode on the joystick. IIRC the green LED must be off to work correctly with Linux.

The buttons are hard-coded in the nav2d_remote package, so if you need to change them, you have to recompile. But as I said, with the F710 this should not be necessary.

The controls are a bit different between simulation (sim_joy) and robot (remote_joy). With sim_joy you hold one of the color-buttons to select a robot and control it with the left stick. With remote_joy you use the left stick to steer and the right stick for forward/backward motion.

CapitalZe, 2018-05-22:
Thanks, I'll check that.

Much appreciated.

closed 2018-05-22 by CapitalZe