skasperski/navigation_2d #53: Should explored-area increase monotonically or not?

s-arora-1987, 2019-04-09:
I have implemented a code that, after irregular time intervals, subscribes to the message published by
mMapPublisher = robotNode.advertise<nav_msgs::OccupancyGrid>(mMapTopic, 1, true);

and uses the data and the metadata in this message to compute the %age of map known (area known/ total area)?

The percentage, sometimes, goes lower than that computed in previous interval. Is it possible in current implementation that %age of known map can vary non-monotonically like 55%, 65%, 60%, 66% ...?

skasperski, 2019-04-10:
It depends on how you calculate the map coverage. Generally, after a loop closure is performed, the size of the map may decrease due to overlapping areas that had been rendered separately before the loop was closed.

s-arora-1987, 2019-04-12:
I calculated the map coverage as (number of cells in occupancy grid that are not -1)/ (total number of cells). I did this calculation for the map learned by each robot individually (data publisher by mMapPublisher in respective nodes), and then I sum up the results. In this case, shouldn't loop closure increase sum of covered area rather than decreasing it?

skasperski, 2019-05-15:
No, Loop-Closures usually decrease the size, because parts of the map are merged.

s-arora-1987, 2019-09-02:
I am trying to partition the area of metadata (info of nav_msgs::OccupancyGrid) of the map received by subscribing to mMapPublisher. However, width and height in the metadata is different from the height and the width in floorplan model in .world file for stage_ros. Is there a way to get output of mapper in dimensions given in .world file?

skasperski, 2019-09-03:
> Is there a way to get output of mapper in dimensions given in .world file?

No, the given height and width are those of the generated map from the mapper. This map grows dynamically with the underlying graph and cannot be given a fixed size.

s-arora-1987, 2019-09-06:
I am computing the current coverage in the map learned by only one robot, not both. When a mapper learns more about environment, with or without merger of two maps, the ratio  (number of cells in occupancy grid that are not -1)/ (total number of cells) should increase. If numerator has changed with the size of returned map, then so has denominator. Am I correct in expecting the ratio to increase?

s-arora-1987, 2019-09-09:
I think I found a reason why I don't see any increase. As you can see in attached image here
[https://drive.google.com/file/d/1gEbViEWcZDgodZc4r4h3mfWjnGMMdh1k/view?usp=sharing](https://drive.google.com/file/d/1gEbViEWcZDgodZc4r4h3mfWjnGMMdh1k/view?usp=sharing)
, robot stops moving on reaching boundary of map, and 'odom frame arrow in rviz' is outside the boundary. What can be the reason for that happening?

skasperski, 2019-09-16:
This looks very much like an issue with OpenKarto's grid generation which I encountered myself before. It happens when walls are exactly perpendicular to the grid. Such walls are sometimes cut off when they appear at the boundary of the map. It happens mostly in simulation and can be avoided by selecting a start position with a heading > 0.

skasperski, 2019-09-16:
> I am computing the current coverage in the map learned by only one robot, not both. When a mapper learns more about environment, with or without merger of two maps, the ratio (number of cells in [...]

No, you should not rely on that. As the map grows, larger unexplored areas will also be added to the map, causing the ratio to possibly drop even by large amounts. I suggest to only count the absolute number of explored cells to monitor the progress of the exploration process.

s-arora-1987, 2019-09-16:
> No, you should not rely on that. As the map grows, larger unexplored areas will also be added to the map, causing the ratio to possibly drop even by large amounts. I suggest to only count the [...]

Q1: Is there a way to measure exploration as a percentage rather than an absolute value? 

Q2: As the map grows and larger areas are added to map, shouldn't both numerator and denominator increase proportionally?

skasperski, 2019-09-17:
Q1: If you know the size of your environment, than you can calculate (explored_area / known_size). If you don't know the size, you cannot calculate any percentage.

Q2: No. Just imagine the robot going into a small corridor after a turn. The explored area will grow very little, but the map will grow in its full width, so the unexplored area will grow much more.

closed 2020-01-14 by s-arora-1987