skasperski/navigation_2d #54: Navigator appears to be ignoring provided .yaml configuration

jotadepicas, 2019-11-22:
Hi!

I'm using nav2d in ROS Kinetic, installed as an ubuntu package (available at /opt/ros/kinetic).
In my launch file I'm configuring Operator and Navigator as follows:

```xml
<node name="Operator" pkg="nav2d_operator" type="operator" >
	<rosparam file="$(find nav2d_tutorials)/param/operator.yaml"/>
	<rosparam file="$(find nav2d_tutorials)/param/costmap.yaml" ns="local_map" />
</node>

<node name="Navigator" pkg="nav2d_navigator" type="navigator">
	<rosparam file="$(find nav2d_tutorials)/param/navigator.yaml"/>
</node>

```
The thing is, Operator complains that "robot" frame is not defined (or at least the error arises when Operator is running, although robot_frame is a Navigator parameter, not sure why):

```bash
"robot" passed to lookupTransform argument source_frame does not exist.
```
 
By looking at the code and Navigator documentation, it appears that "robot" is what the robot_frame value defaults when it's not provided (for example, https://github.com/skasperski/navigation_2d/blob/master/nav2d_operator/src/RobotOperator.cpp#L16)

I noticed that neither operator.yaml, navigator.yaml nor costmap.yml specify "robot_frame", so I went ahead and added it (at first to navigator.yaml, then everywhere just in case) and nothing happened. Moreover, I see that the only yaml in the tutorials that specifies robot_frame is ros.yaml, which is not passed as argument in any of the examples (https://github.com/skasperski/navigation_2d/blob/master/nav2d_tutorials/param/ros.yaml).

So, I edited ros.yaml so robot_frame would point to my frame, which is called base_footprint, and I even passed ros.yaml to navigator as well, but the error persists. The only thing that ended up working was creating a static tf and creating "robot" based on "base_footprint":

```xml
<node pkg="tf" type="static_transform_publisher" name="nav2d_hack"
args="0 0 0 0 0 0 base_footprint robot 100" />
```

In other words, no matter what I do, the code seems to be ignoring the robot_frame value passed in yaml, and defaults to look for it with the name "robot", so I created a frame name "robot" fot it to finally find it, but this doesn't look right.

Why could the code be ignoring the yaml configuration?

I saw in some other questions that you mentioned it was not adviced to mix nav2d installed as ubuntu package and packages as source in catkin workspace (my packages are source in catking_ws). Could this be related to that? Do I need to get nav2d as code into my workspace for the yaml configuration to work?

Thanks!

skasperski, 2019-11-22:
The frames are defined at top level in a separate file called [ros.yaml](https://github.com/skasperski/navigation_2d/blob/master/nav2d_tutorials/param/ros.yaml). This is because all nodes should use the same frame and topic names. Adding the frame to the navigator.yaml doesn't help, because this one is loaded into the navigator's namespace.

jotadepicas, 2019-11-22:
Thanks for your prompt reply. Oh I see, it's a namespaces issue. I completely missed ros.yaml being a param in the global namespace (I was looking at params for each node individually). It confused me that I saw Navigator/base_frame being passed, and I couldn't understand why it was being ignored.

Let me check if this works and I will come back here and confirm.

jotadepicas, 2019-11-22:
Yes, that was it, I was missing ros.yaml as a top-level param. I'll close the issue. Thanks again!

closed 2019-11-22 by jotadepicas