skasperski/navigation_2d #55: Using two LIDARs with nav2d_karto

DannySauval, 2019-12-04:
Hi,

I'm using SLAM Karto with two LIDARs mounted at opposite sides of a rectangular robot. I'm currently using [Ira laser tools](https://github.com/iralabdisco/ira_laser_tools) to merge the data from the two sensors. I feed the output to Karto. 

This runs fine, but not great. I don't think this is the most efficient solution, and Ira laser tools sometimes generates wrong data (I'm still trying to figure out why). I would like to know if there is a better way to use two lidars with SLAM Karto. I tried running two Mapper nodes making use of karto_in and karto_out, but I can't get it to work. Is there a better way to feed karto with two LIDARs ?

Thanks !

skasperski, 2019-12-04:
> I'm using SLAM Karto with two LIDARs mounted at opposite sides of a rectangular robot. I'm currently using [Ira laser tools](https://github.com/iralabdisco/ira_laser_tools) to merge the data from [...]

I think this is actually the most efficient and stable solution. It creates a virtual 360° scan, which is very good for scan matching and should give you the best results. Make sure that your scanners are well calibrated, e.g. their relative position is very accurate. If you obtain poor results, I assume it is because of poor alignment of the two sensors.

> I would like to know if there is a better way to use two lidars with SLAM Karto. I tried running two Mapper nodes making use of karto_in and karto_out, but I can't get it to work.

While this should work theoretically, I cannot recommend this solution. Multiple mapper instances are meant to be deployed to multiple robots. Having two instances on one robot is a very inefficient approach.

> Is there a better way to feed karto with two LIDARs ?

There should be a third way, but it is more difficult and would require to change the implementation of the mapper node. You could add two different LaserScanners to Karto and use them both for mapping. The node would need two subscribe two sensor_msgs/LaserScan and add them individually to the map. However this will likely produce worse results, because scans of the front and back laser cannot be matched properly against each other. 

So if nothing stands against it, fusing the scans before adding them to the mapper seems the best solution to me.

DannySauval, 2019-12-13:
Hi !
Thank you for your answer. I'll trust your experience and go with Ira Laser Tools. I kind of put this issue aside for now but I'll try to debug what's wrong in my code.
Have a nice day !

closed 2019-12-29 by DannySauval