skasperski/navigation_2d #66: Question about TF between Stage simulator and nav2d_karto

bairuofei, 2023-03-08:
Hi, when running nav2d for navigation, I find it is difficult to understand how frame_id in Stage and in nav2d_karto are related.

1. Stage simulator publishes the message "/base_pose_ground_truth" with a frame_id "odom". The "odom/" frame is fixed according to .world file, typically being the initial position of the robot when setting localization as "odom" in .inc file.
2. nav2d_karto publish grid map "/map" and robot pose "/others" with a frame_id "map". 

It seems that in rviz, the frame "odom" and "map" is closer each other, and changes slightly as SLAM operates. After checking the rqt_graph, I find the /Mapper node only subscribe the topic "base_scan" from the /Stage node. It seems there is no direct relationship between the two frame "odom" (in Stage) and "map" (in Karto SLAM). May I know at the very beginning how these two frames are related to each other? 

[image]

skasperski, 2023-03-08:
All frame-handling is done via topic /tf, if you enable tf in rqt_graph you will see that the mapper is connected via tf as well.

The mapper publishes the transform map->odom, so that the odometry frame slowly moves away from the origin, this is the expected behavior. (This is the correction of the odometry, which is calculated from the underlying SLAM)

You can use "rosrun tf view_frames" to generate a PDF with all frame information in the system.

bairuofei, 2023-03-09:
That's right, thank you for the explanation. The problem is really confusing and I also encounter a similar question [here.](https://answers.ros.org/question/306403/some-problem-with-global-localizaton-using-stage-simulation/) Could you please check whether the following statements are right?

1. The "map" frame is set equal to "odom" frame at the very beginning.
2. When SLAM node receives a new "/base_scan", the scan is first positioned following the tf chain: map -> offset -> odom -> base_laser_link.
3. SLAM node registers the "/base_scan" in current map, and gets an optimized pose estimation, which is different from the odometry estimation. The frame "offset" is adjusted to mitigate the difference, so that the "base_laser_link" is aligned with the optimized pose estimation given by SLAM node. Accordingly, the frame "odom" is slightly moving around the frame "map" because of the tf chain: map -> offset -> odom.
4. So, to recover the ground truth pose of the robot, and the odometry pose of robot (both with frame_id being "odom", comes from Stage), one can just simply change the frame_id of these two topics as "map", which is actually the inital origin of the  "odom" frame.

Am I right? Thank you for your time!

closed 2023-03-13 by bairuofei